dmadison/Adalight-FastLED #20: Flashing randomly

jebbett, 2020-03-30:
Another user and I have been getting flashing while using with Prismatik, adding the below to the script solves this issue, in case anyone else has the same issue

#define FASTLED_ALLOW_INTERRUPTS 0

dmadison, 2020-03-30:
Could you please provide more information? Does the flashing happen constantly or intermittently? Is the flashing a specific color? Do you have a way to reliably trigger it?

Please also provide the following information about your hardware and software setup:
* What board are you using?
* How many LEDs are you using?
* What are your settings for this script?
    * Baud rate
    * LED type
    * Optional settings enabled
* What is your "grab rate" in Prismatik?

jebbett, 2020-03-30:
Flashing is random, can be on a static screen, although less frequently, flashing is white light from a random LED.
Running a Wemos D1 Mini, 68 WS2812B LEDs.
115200 baud, nothing optional.
I've seen issues in the past with FastLED, at least on the D1, adding the line I posted fixes the flashing, so don't need a fix just letting you know.

dmadison, 2020-03-31:
It's likely that your "grab rate" in Prismatik is too low (capturing too frequently) for your selected settings, and the flashing you're seeing is due to serial data being sent while the current frame is being written. The ESP8266 boards don't disable interrupts with FastLED when writing to WS2812B's like on most boards, so serial interrupts can disturb the LED timing and cause issues.

My bet is that the `#define FASTLED_ALLOW_INTERRUPTS 0` setting is masking the real issue, which is that you're trying to push data too quickly. I would recommend either lowering your grab rate in Prismatik (higher ms number), bumping up your baud rate, or both. See this article for reference: https://www.partsnotincluded.com/calculating-adalight-framerate-limits/

jebbett, 2020-03-31:
Makes sense will give it a go, it's only at 17 FPS, but will try upping the baud rate

dmadison, 2020-03-31:
17 FPS (60 ms) or 17 ms?

jebbett, 2020-03-31:
17FPS (60ms)

jebbett, 2020-04-01:
Changing the Baud seems to make no difference, I've re-added the line I've suggested and bumped it up to 30 FPS and it's working fine.. go ahead and close, it seems to be working fine with the line added, looks great on films and games on my ultrawide, so thanks for your code!

closed 2020-04-02 by dmadison