dmadison/Adalight-FastLED #21: fastlights rainbow Mode in Ambibox

KiWy86, 2020-04-02:
is it possible to integrate a rainbow mode into the arduino code?  as normally available in the fastLED.  or is that only possible by creating a profile in ambibox or via a dll?

dmadison, 2020-04-03:
Sure it's possible, you'd just have to figure out how you want to trigger it. Then you can add it like you would to any other Arduino program.

Depending on what you're doing you may be better off adding a "rainbow" effect as a feature of the program you're using on the PC. I do not use or like ambibox so I can't help you there, but Prismatik has a socket-based API you can use for setting LED colors from a script.

dmadison, 2020-04-17:
Assuming solved.

closed 2020-04-17 by dmadison