dmadison/Adalight-FastLED #23: Not working for me

marcosruiz, 2020-05-21:
I bought [this kit](https://es.aliexpress.com/item/32964959328.html?spm=a2g0s.9042311.0.0.2acd63c0pU1i0x with WS2812B) (180 leds). It worked with the original program but I tried to install your code following your instructions and it is not working.

I edited this lines:

```
static const uint16_t 
	Num_Leds   =  180;   

#define COLOR_ORDER  GRB 
 ```

The controller is a Arduino UNO chinese clone.

The FastLED library that I use is "FastLED version 3.001.003".

The ambibox shows: "Device status: connected" and "Adalight: connected" with a "capture speed: 20 FPS" which looks great but LEDs do not turn on...

Do you know what could be the problem? I can suppose that the led is not the WS2812B but I dont know how to check that.

dmadison, 2020-05-22:
My first guess would be that [the output pin in the program](https://github.com/dmadison/Adalight-FastLED/blob/9a380a57c3f3fc342a64f72c0404b0e90cf90bea/Arduino/LEDstream_FastLED/LEDstream_FastLED.ino#L33) does not match the data pin on your hardware. I would open up the hardware and check that first. If it's a custom circuit board youll likely have to follow the traces to the microcontroller and look up the datasheet to find the pin (sorry).

I would also recommend using Prismatik, at least for testing. I've seen lots of strange issues reported with Ambibox.

marcosruiz, 2020-05-22:
Wow, thanks for your quick response. **I am not using any pin**, I am using the power connection to send the RGB info to the leds:

This is the schema:
PC ---> USB cable ---> Chinese Arduino UNO ---> Power cable ---> String leds

Also the string leds are feeded with Power with a power supply.

Has the power connection something like a PIN_NUMBER? Maybe this is not a question to do in this thread but I am not very familiarized with Arduino.

dmadison, 2020-05-22:
The power connection on an Arduino Uno cannot be used as an I/O pin.

Looking at that product page you linked, it appears as though they're just using a barrel connector for the data connection to the strips. You will still need to figure out what digital I/O pin that barrel connector is attached to.

marcosruiz, 2020-05-22:
I finally understood it. The cable is connected from PIN 8 to the barrel connector in the bottom of the UNO. You were right!!. Thank you, now is working!

closed 2020-05-22 by marcosruiz