dmadison/Adalight-FastLED #27: Arduino Nano - baud rate

kovapatrik, 2020-12-01:
Hi!

I tried to change the baud rate to 460800 for having more FPS, but then the Nano couldn't control the LEDs. I have 103 LEDs and with the baud rate set to 153600, it's working.

dmadison, 2020-12-01:
That's not an issue with the firmware, that's an issue with the hardware on your microcontroller and what baud rates it supports. If you're using a knockoff Nano with a CH340G instead of an FT232RL, try using either 250,000 or 500,000 baud. Otherwise I would recommend testing the baud rate and serial interface using a terminal program standalone (i.e. not using the Adalight firmware) until you find a fast baud rate that works on your hardware.

kovapatrik, 2020-12-01:
Ah okay, thanks for your help!

closed 2020-12-01 by kovapatrik