dmadison/Adalight-FastLED #30: Can't get to work using Prismatik on WS2812 (non-B) and Elegoo Mega 2650 R3

Duskofdawn, 2021-03-11:
I've been messing around with this for a little bit in order to get it working with my particular setup and haven't found a way to get it to work. I've got the script set up with the following settings:

```
// --- General Settings
const uint16_t 
	Num_Leds   =  144;         // strip length
const uint8_t
	Brightness =  255;        // maximum brightness

// --- FastLED Setings
#define LED_TYPE     WS2812  // led strip type for FastLED
#define COLOR_ORDER  GRB      // color order for bitbang
#define PIN_DATA     7        // led data output pin
// #define PIN_CLOCK  7       // led data clock pin (uncomment if you're using a 4-wire LED type)

// --- Serial Settings
const unsigned long
	SerialSpeed    = 115200;  // serial port speed
const uint16_t
	SerialTimeout  = 0;      // time before LEDs are shut off if no data (in seconds), 0 to disable
```

I've got Prismatik set up as an Ardulight running baud 115200 on COM3 (the port the board is running on.) The board itself has the L light solid and RX light blinking rapidly, I've tried both under screen grabbing and mood light mode. The strip itself is being powered off the 5V line of a 650W PSU. I can run blink.ino with the following (shortened) code just fine:

```
#include <FastLED.h>
#define NUM_LEDS 144
#define DATA_PIN 7
CRGB leds[NUM_LEDS];
void setup() {FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, NUM_LEDS);}
void loop() { 
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

Once reuploading the LEDstream_FastLED code to the board, the TX light blinks once per second, serial monitor shows Ada. As soon as I start Prismatik up, TX light goes off, RX light starts flashing rapidly, but nothing actually happens with the lights (on the strip) themselves.

I don't believe the issue to be an issue with your software, but I've run through everything I can think of and as far as I can tell, everything is functioning...I just don't get the expected result. I know this requires you to take free time to answer. Any help you can give will be appreciated.

dmadison, 2021-03-11:
>I've got Prismatik set up as an Ardulight

This is Adalight, not Ardulight. Make sure you have the right setting selected.

Duskofdawn, 2021-03-11:
> > I've got Prismatik set up as an Ardulight
> 
> This is Adalight, not Ardulight. Make sure you have the right setting selected.

It's working. It had to have been some combination of that and something else I was doing. Thank you for the assistance and sorry for the time taken.

closed 2021-03-11 by Duskofdawn