dmadison/Adalight-FastLED #34: Not an issue

jordanadania, 2023-01-10:
I don't know where "discussion" is.

I was just curious how I could use this with X and Y coordinate mapped LEDs? Like a Fibonacci layout.

I tried: `ledsRaw[myMap[outPos++]] = c; // Issue next byte`
But that didn't work for me.

Thanks in advance.

dmadison, 2023-01-10:
Hi Jordan. Don't worry, you're in the right place.

The code was not written with that functionality in mind. The Adalight protocol delivers bytes as a single RGB datastream. The code treats the `CRGB` objects as a singular byte-sized buffer in RGB order and stores the bytes one at a time as they come in.

To do what you're suggesting, you would need to add a 3-byte counter and buffer within the data storage function. Every time you count three bytes, store the resultant RGB data into the LED buffer using your XY function. If I had written the code today it probably would be counting the color channels and doing that *anyways*, but alas...

I'm not sure what your use case is (Fibonacci LEDs around a monitor doesn't make much sense to me), but you might also consider configuring whatever program is sending the data to set the layout there instead.

jordanadania, 2023-01-17:
Incredible. Thank you so much!

closed 2023-01-17 by jordanadania